NeBula-Autonomy/LOCUS #48: Can not start LOCUS in any way, please help.

Cristian-wp, 2022-07-22:
Hello,
I am trying to run LOCUS on my dataset. My dataset consists in a rosbag with imu (vectornav vn100), 3D lidar (VLP16), odometry topic (from a visual slam, type nav_msgs/Odometry) ad a camera image.

When I try to run locus alone and start the rosbag, I get this output:

```
roslaunch locus locus.launch
... logging to /var/log/extars/ros/6dabe0b2-09a6-11ed-bbec-54b203083f30/roslaunch-extars-28081.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
started roslaunch server http://extars:45129/

SUMMARY
========

PARAMETERS
 * /extars/body_filter/max_x: 0.6
 * /extars/body_filter/max_y: 0.5
 * /extars/body_filter/max_z: 1.8
 * /extars/body_filter/min_x: -1.0
 * /extars/body_filter/min_y: -0.5
 * /extars/body_filter/min_z: -1.6
 * /extars/body_filter/rotation: 0.0
 * /extars/extars/robot_description: <?xml version="1....
 * /extars/extars/tf_prefix: extars
 * /extars/locus/b_adaptive_input_voxelization: True
 * /extars/locus/b_add_keyframes_enabled: True
 * /extars/locus/b_debug_transforms: False
 * /extars/locus/b_enable_computation_time_profiling: True
 * /extars/locus/b_enable_msw: True
 * /extars/locus/b_integrate_interpolated_odom: True
 * /extars/locus/b_is_flat_ground_assumption: False
 * /extars/locus/b_pub_odom_on_timer: False
 * /extars/locus/b_publish_xy_cross_section: True
 * /extars/locus/b_run_with_gt_point_cloud: False
 * /extars/locus/b_sub_to_lsm: False
 * /extars/locus/b_verbose: False
 * /extars/locus/box_filter_size: 20
 * /extars/locus/buffers/imu_buffer_size_limit: 100
 * /extars/locus/buffers/odometry_buffer_size_limit: 100
 * /extars/locus/data_integration/mode: 0
 * /extars/locus/filtering/decimate_percentage: 0.9
 * /extars/locus/filtering/decimate_percentage_open_space: 0.93
 * /extars/locus/filtering/extract_features: False
 * /extars/locus/filtering/grid_filter: False
 * /extars/locus/filtering/grid_res: 0.2
 * /extars/locus/filtering/outlier_filter: False
 * /extars/locus/filtering/outlier_knn: 10
 * /extars/locus/filtering/outlier_std: 1.0
 * /extars/locus/filtering/radius: 0.15
 * /extars/locus/filtering/radius_filter: False
 * /extars/locus/filtering/radius_knn: 3
 * /extars/locus/filtering/random_filter: False
 * /extars/locus/frame_conversions/b_convert_imu_to_base_link_frame: True
 * /extars/locus/frame_id/artifacts_in_global: True
 * /extars/locus/frame_id/base: extars/base_link
 * /extars/locus/frame_id/bd_odometry: extars/odom
 * /extars/locus/frame_id/fixed: extars/map
 * /extars/locus/frame_id/imu: extars/imu_link
 * /extars/locus/gt_point_cloud_filename: /home/costar/data...
 * /extars/locus/icp/corr_dist: 1.0
 * /extars/locus/icp/enable_timing_output: False
 * /extars/locus/icp/iterations: 20
 * /extars/locus/icp/max_rotation: 1.0
 * /extars/locus/icp/max_translation: 1.0
 * /extars/locus/icp/num_threads: 4
[... 212 lines cut ...]
    load_entry_point('tmuxp==1.3.5', 'console_scripts', 'tmuxp')()
  File "/home/ctrazzi/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 479, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/ctrazzi/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2861, in load_entry_point
    return ep.load()
  File "/home/ctrazzi/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2465, in load
    return self.resolve()
  File "/home/ctrazzi/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2471, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python3/dist-packages/tmuxp/__init__.py", line 20, in <module>
    from . import config, util, cli
  File "/usr/lib/python3/dist-packages/tmuxp/cli.py", line 550, in <module>
    help='Like -2, but indicates that the terminal supports 88 colours.')
  File "/home/ctrazzi/.local/lib/python3.6/site-packages/click/decorators.py", line 223, in decorator
    _param_memo(f, ArgumentClass(param_decls, **attrs))
  File "/home/ctrazzi/.local/lib/python3.6/site-packages/click/core.py", line 2905, in __init__
    super().__init__(param_decls, required=required, **attrs)
  File "/home/ctrazzi/.local/lib/python3.6/site-packages/click/core.py", line 2030, in __init__
    param_decls or (), expose_value
  File "/home/ctrazzi/.local/lib/python3.6/site-packages/click/core.py", line 2941, in _parse_decls
    "Arguments take exactly one parameter declaration, got"
TypeError: Arguments take exactly one parameter declaration, got 2.
```
Can you please help me to make it work?
Can it be realated to the no dense pointcloud?

femust, 2022-07-24:
Hey @Cristian-wp 

Thanks for testing and trying out our solution!

At the first sight, it seems that it's related to your pc setup and it's weird that sometimes you run python2 and python3. It may be causing some problems.

What I suggest you could do, without fixing the local problems, is go on branch `feature/docker` [here](https://github.com/NeBula-Autonomy/LOCUS/tree/feature/docker) and follow [README steps](https://github.com/NeBula-Autonomy/LOCUS/tree/feature/docker#dockerfile). If you run docker, by default is connected with "your host network" so if you run `roscore` locally the docker sees that and you may run locus. Let me know if that helps you!

Cristian-wp, 2022-07-26:
I have format my pc and now it start.

closed 2022-08-02 by femust