NeBula-Autonomy/LOCUS #54: how to get wio_ekf

linwk20, 2022-09-27:
Hi, Thanks for your sharing!
However, when I check the test bag, I wonder how can I get the wio_ekf ? since I only got wheel odometry in my dataset.

BenjaminMorrell, 2022-09-27:
Wheel odometry should work fine in place of WIO_EKF. That topic was produced from a separate sensor fusion package that is on open-sourced at the moment (it combines IMU and wheel odometry). 

I think just wheel odometry would be ok - your roll and pitch isn't going to be very helpful, but if you have enough lidar features that should be ok. 

Otherwise, the general approach is to use the IMU for roll and pitch, wheel odom for x, y, z, yaw - but fused through an ekf I believe.

linwk20, 2022-09-27:
Thanks for your quick reply! I get it!

closed 2022-09-27 by linwk20