NeBula-Autonomy/LOCUS #56: MDC module problem

linwk20, 2022-09-27:
Hi, Thanks for sharing this great work!
I am now trying to use LOCUS on my own Lidar dataset. However, as said in the repo, there is no MDC module provided.
I want to know if I just add a deskew module from another repo, can I fix this problem? Or is there anything else to take care of?

Thanks!

BenjaminMorrell, 2022-09-27:
Thanks for using the repo! 

Yeah, that would be one way to approach it - if you find another good deskewing module it would be great if you could post a link here for the benefit of others. 

Another consideration is that if your system is moving relatively slowly (in linear and rotational velocities), then just using the point-clouds as they are might be ok. 

@femust - have you looked at what it would take to release our velodyne driver MDC?

linwk20, 2022-09-28:
It seems LIO-SAM 's imageProjection.cpp unit can be applied easily. I will try, thanks for your quick reply!

https://github.com/TixiaoShan/LIO-SAM

BenjaminMorrell, 2022-10-06:
Great! Let us know how you go!