NeBula-Autonomy/LOCUS #57: catkin build locus failed

MoonSheepJx, 2022-11-01:
First of all, thank you very much for a great job, but I am having some problems compiling and would appreciate your help.
After install all required tools (pcl_ros catkin_tools...),  i taked locus and common_nebula_slam unzip into catkin_ws/src , my folders are layered as follows:
catkin_ws
       src
            core_msgs
            docker
            frontend_utils
            geometry_utils
            locus
            ......
and i **catkin build locus** at catkin_ws/, some errors appeared:
[image]

moreover, my environment：Ubuntu18.04  ROS melodic PCL1.8  BOOST1.65
I am looking forward to your reply，thanks

MoonSheepJx, 2022-11-01:
It shows missing files, i search "BodyFilterConfig.h" in my folder, but not found.

xchwang1998, 2022-11-01:
I have the same problem, and I couldn't find "BodyFilterConfig.h".

femust, 2022-11-01:
and if you enforce the ```catkin build``` again, the same happens correct? (without cleaning space)

MoonSheepJx, 2022-11-01:
i have build success！

I searched for the project and could not find the file. I thought it was generated automatically by the code. This also happens in many projects.
I do like this: when error, do not clean space and enforce build again
specifically，i **catkin build locus -j2** to control memory, and above error occurred, then  keep trying to compile with more or fewer threads
like **catkin build locus -j4** or **catkin build locus -j1** (i have only 8 threads)
then build successfully although some warning. 
[image]
Thank you for your tips and help. I will test later

femust, 2022-11-01:
I guess some dependencies are not included properly, then. Let me reopen this thread and review it so we get rid of it for good :)