NeBula-Autonomy/LOCUS #58: How to test LOCUS?

xchwang1998, 2022-11-01:
I use the command 
> roscore
then 
> tmuxp load {path_to_locus}/LOCUS/tmuxp_config/run_locus.yaml
( the path in run_locus.yaml have been revised to myself)

[image]

The result are as following:
[image]

This is not same with demo.

femust, 2022-11-01:
Have a look that you are on the first tab ```1: record_at_end``` . So click ```locus_datt0:locus``` to go to the zero tab.

[image]

I think in older versions of tmuxp the scripts are run in a weird way, while on the newer version it starts from the "zero tab"

Let me know if that worked for you

xchwang1998, 2022-11-02:
> Have a look that you are on the first tab `1: record_at_end` . So click `locus_datt0:locus` to go to the zero tab.
> 
> [image]
> 
> I think in older versions of tmuxp the scripts are run in a weird way, while on the newer version it starts from the "zero tab"
> 
> Let me know if that worked for you

I click `locus_datt0:locus`, and no response.
[image]

The version of tools on my computer as following:
> tmux: 2.6  
> tmuxp: 1.3.5

[image]

It's version is really old. What version do you recommend?

MoonSheepJx, 2022-11-02:
I have the same problem now.

xchwang1998, 2022-11-02:
I install them by:
> sudo apt install tmux
> sudo apt install tmuxp

Should I install them through source code?

femust, 2022-11-02:
https://help.sourcelair.com/terminal/working-with-multiple-tabs/ try this

I think you guys use ubuntu 18?

xchwang1998, 2022-11-02:
Yes my system is ubuntu 18.04. I switch to ‘locus_datt0:locus’ by hit Ctrl + B and then 0.

Here have some errors, but I had source the workspace before tumxp load XXX.yaml.
[image]

femust, 2022-11-02:
put in ~/.bashrc the path to the workspace

```
source ~/workspace/locus_ws/devel/setup.bash
```
in your case, or something like this.

xchwang1998, 2022-11-02:
> put in ~/.bashrc the path to the workspace
> 
> ```
> source ~/workspace/locus_ws/devel/setup.bash
> ```
> 
> in your case, or something like this.

That's OK! Thanks! By the way, why do we add this command in ~/.bashrc?
> ```
> source ~/workspace/locus_ws/devel/setup.bash
> ```
I try in locus_ws use this command before 'tumxp load' and it didn't work.
> ```
> cd ~/workspace/locus_ws
> source devel/setup.bash
> tmuxp load {path_to_locus}/LOCUS/tmuxp_config/run_locus.yaml
> ```