NeBula-Autonomy/LOCUS #59: How to running LOCUS on my own dataset

MoonSheepJx, 2022-11-02:
I want to running LOCUS on other datasets, my datasets format is `.bag`
`rosbag play xxx.bag` then publish the topic
according to README,  it seems that only `run_locus.yaml`  and `locus.launch` need to be modified.
i wonder to know what i need to modify can connect the datasets and system, such in other SLAM (just need to modify the topic in config)

i think need to modify the remap in `locus.launch`
[image]
and data path in `run_locus.yaml`
[image]
but not success,  provide some help please.

Moreover, my datasets have LiDAR data and IMU data with .bag format, but there is no data of odometry(wheel odometry)(input).

xchwang1998, 2022-11-03:
同学你好，能否加个联系方式交流一下，微信13032462319

AcamelQ, 2023-06-02:
i met the same issue. can anybody help me with this?