NeBula-Autonomy/LOCUS #64: LOCUS maximum accuracy mode for GT trajectory

Telios, 2023-11-22:
Hi,
I want to create a ground truth trajectory with LOCUS. I already managed to sucessfully load a prior scanned map into LOCUS and record the odometry msgs for further use with evo. Are there some parameters other than the b_run_with_gt_point_cloud and b_add_keyframes_enabled that have to be changed for LOCUS to run at maximal accuracy sacrificing real-time, as mentioned in [the odometry dataset](https://github.com/NeBula-Autonomy/nebula-odometry-dataset/blob/main/pages/dataset.md#ground-truth)?
Thank you in advance!

BenjaminMorrell, 2023-11-22:
Thanks for your question Telios. The two settings you mention are the most important. For the reference of others looking to do the same:
- `b_run_with_gt_point_cloud` - set to true [here](https://github.com/NeBula-Autonomy/LOCUS/blob/main/locus/config/lo_settings.yaml#L39).
- `gt_point_cloud_filename` - your filename for the prior map (`.pcd` file) [here](https://github.com/NeBula-Autonomy/LOCUS/blob/main/locus/config/lo_settings.yaml#L40C1-L40C24)
- `b_add_keyframes_enabled` - set to false [here](https://github.com/NeBula-Autonomy/LOCUS/blob/main/locus/config/lo_settings.yaml#L43C1-L43C24)

For the tuning mentioned in the readme, it is a bit of an iterative process for the specific dataset you are working with. 
Some tuning factors to consider include. 

1. Try increasing the number of points coming out of the filter. The relationship here between more points and better results wasn't that direct from our experience, but if you have a nice ground truth scanned map, then I would expect more points to help. Set [this](https://github.com/NeBula-Autonomy/LOCUS/blob/main/locus/config/lo_settings.yaml#L85) to a higher number. 
2. Adjust the GICP parameters for the scan-to-map step [here](https://github.com/NeBula-Autonomy/LOCUS/blob/main/point_cloud_localization/config/parameters.yaml). I would start with increasing [iterations](https://github.com/NeBula-Autonomy/LOCUS/blob/main/point_cloud_localization/config/parameters.yaml#L23) so that never limits it. Depending on your scan resolution and filtering settings of the scans, you may want to decrease `corr_dist` [here](https://github.com/NeBula-Autonomy/LOCUS/blob/main/point_cloud_localization/config/parameters.yaml#L20C2-L20C2), so points need to be closer to be considered matches (avoiding potentially incorrect matches). You might consider looking at decreasing [tf_epsilon](https://github.com/NeBula-Autonomy/LOCUS/blob/main/point_cloud_localization/config/parameters.yaml#L16) if that is causing exits from the optimization, but I don't think that is likely to make a large difference. 
3. Depending on your data quality, remove any data integration (set [this](https://github.com/NeBula-Autonomy/LOCUS/blob/main/locus/config/lo_settings.yaml#L32) to 0) - to just prioritize the lidar data. If you trust your calibration to IMU or other odometry source, then leave it on. We tended to bias toward the map being the source of truth, which is only linked with lidar - hence we ignored other sensor sources for the ground truth generation. 
4. Definitely try running at a slower speed (`rosbag play filename.bag -r 0.1` for instance)
5. You might want to increase the lidar queue size ([here](https://github.com/NeBula-Autonomy/LOCUS/blob/main/locus/config/lo_settings.yaml#L18C15-L18C15)) depending on the rate that you are running, to make sure you don't drop any scans (it will just delay the output pose, which is fine for this). 

Other than these, there may be more fine adjustment of the GICP parameters in [here](https://github.com/NeBula-Autonomy/LOCUS/blob/main/point_cloud_localization/config/parameters.yaml), but hopefully you don't need that.

Telios, 2023-11-23:
Thank you very much for this detailed answer! This will definetly improve the accuracy of my GT trajectory. You guys made an awesome project!

closed 2023-11-23 by Telios